bassamanator/Sovol-SV06-firmware #6: Print Cancel Issues

s3anami, 2023-03-10:
I am using this firmware on my Creality Sonic Pad, and occasionally have issues with print cancellations. Occasionally when I cancel a print it doesn't truly fully cancel it. For example, I can the print, cancel it on the pad, clear the bed and attempt to reprint, the printer will attempt to resume printing instead of restarting for some reason. I have to power cycle the printer to clear it.

Another issue I had were the mesh point settings. They were also off for me compared to the official and some other cfg set ups ( too low on the build plate, y value too low, and x values too low as well).

bassamanator, 2023-03-10:
Although I haven't made any changes to any of these macros recently, are you up to date with this repo? I cancelled some prints just yesterday and didn't have any issues.

I don't know what the sonic pad does. Can you start and cancel the print using the Mainsail or Fluidd interface?

bassamanator, 2023-03-10:
> Another issue I had were the mesh point settings. They were also off for me compared to the official and some other cfg set ups ( too low on the build plate, y value too low, and x values too low as well).

I don't follow you here, please elaborate. You need to adjust your z_offset as well, this is all very clearly explained in the `Initial Steps` section.

s3anami, 2023-03-10:
yup i did, i was commenting on the default values, don't need to be rude. Good

closed 2023-03-10 by s3anami