bassamanator/Sovol-SV06-firmware #27: Y axis out by 25mm

btwo5, 2023-04-12:
Hi, i know its not the cable as i ran fine on stock and hills firmware

ive flashed to klipper and whenever i home via the menu or g28 it doesnt come all the way to the front of the y axis.
when trying to bed mesh the very last row grinds as it cant go further back

any help would be appreciated as ive spent hours on this now

thanks

bassamanator, 2023-04-12:
You need to start reading from [here](https://www.klipper3d.org/TMC_Drivers.html?h=driver_sgthrs#configure-printercfg-for-sensorless-homing), all the way to [Update printer.cfg with sensitivity value](https://www.klipper3d.org/TMC_Drivers.html?h=driver_sgthrs#update-printercfg-with-sensitivity-value). The idea is to determine the correct sensitivity value for sensorless homing for the y axis for your setup. Incidentally, I happen to have gone through this all over again just yesterday, it's very easy and quick. However, if you like you can use the sensitivity value set by Sovol and make the following changes:
```
[tmc2209 stepper_y]
...
driver_SGTHRS: 68
...
```
Remember to be ready to hit the emergency stop at any time.

bassamanator, 2023-04-12:
Please see my comment in the discussion you created.