bassamanator/Sovol-SV06-firmware #43: [BUG] No purge line SV06 Plus?

LsHallo, 2023-04-22:
**Describe the bug**
When using the PRINT_START and PRINT_END macro no purge line is printed resulting in inconsistent first layers.

**To Reproduce**
1. Set Cura pre-print to "PRINT_START"
2. Set Cura post-print to "PRINT_END"
3. Print the sliced G-Code
4. Observe that no form of purge line is printed before starting the print.

**Expected behavior**
I would expect the macro to print the purge line. This line would be in the cura pre-print code previously but with Klipper it is common practice to replace the whole pre-print block with just "PRINT_START".

**Screenshots**
[image]

**What kind of device are you using to host Klipper?**
 - OS: Debian @ BigTreeTech CB1
 - Version: Klipper 011b4e3

**How did you install Klipper?**
 - KIAUH

bassamanator, 2023-04-22:
Change your Cura start code as follows:
```
PRINT_START BED={material_bed_temperature_layer_0} HOTEND={material_print_temperature_layer_0} CHAMBER={build_volume_temperature}
PURGE_LINE
```

bassamanator, 2023-04-22:
Re-open if the need arises.

closed 2023-04-22 by bassamanator