bassamanator/Sovol-SV06-firmware #49: Cannot seem to get this method working? please help, where am I going wrong?

MisteroUK, 2023-04-26:
Hi,

So I am trying to get this to work with my sv06, SKR mini E3 V3 and im struggling at the section labelled "Download Klipper Configuration"

The firmware flashed fine, I installed a mainsail klipper image onto a raspberry pi via pi manager but unfortunately this is as far as I can get things working.

Your instructions under "Download Klipper Configuration" say to delete everything in the ~/printer_data/config folder which I have done but the files you're suggesting to download don't match up? 

you said to put everything from the folder "Sovol-SV06-firmware-master" into the now empty ~/printer_data/config folder but with the download there is no folder named "Sovol-SV06-firmware-master". If im honest I have no idea what this folder should look like or hold so any help for a newbie to klipper would be really helpful. 

If you could give a newbie / simple terms of exactly what should be in that folder would be great as every time I try it ..... everything breaks and nothing will work? klipper goes off line and won't start.

Thanks

Steve

bassamanator, 2023-04-26:
I've edited the instructions. Please try again, [skr-mini-e3-v3](https://github.com/bassamanator/Sovol-SV06-firmware/tree/skr-mini-e3-v3).

Keep updating this thread.

MisteroUK, 2023-04-26:
Thank you ;-) so just to get this right you delete everything out the config folder including the link directories and then copy the entire contents of the git folder download in this config folder? I know im a newbie to klipper but there are a lot of files and folders in there that don't seem relevant?

Im sure im missing something here but its still not making sense lol.

bassamanator, 2023-04-26:
That's right, everything inside `~/printer_data/config`.

The skr branch is not as robust with the instructions because I have yet to install mine. The master branch on the other hand has been tried and tested countless times, have faith in those instructions.

bassamanator, 2023-04-26:
It costs nothing to try again, start again [Download Klipper Configuration](https://github.com/bassamanator/Sovol-SV06-firmware/tree/skr-mini-e3-v3#download-klipper-configuration).

Follow and read the steps exactly.

bassamanator, 2023-04-26:
Also, you mentioned git earlier for some reason, if you're downloading the zip, then git is not at play.

MisteroUK, 2023-04-26:
I am downloading the zip as mentioned then uploading the entire contents of that but the printer is def not performing as it should, hot end crashes into the bed, y axis doesn't homeless sense (jumpers are installed on the board) I stopped at this point as the printer was going to get damaged. looks like the config isn't working or being read correctly.

bassamanator, 2023-04-26:
A `firmware_restart` is required after you edit `printer.cfg`. Try that, or the mainsail option does the same thing.

The enclosure fan behaviour can be changed, should turn off after 60 seconds.

MisteroUK, 2023-04-26:
tried it but for some reason its still not working? very strange, now y axis has completely died and will not respond at all... oh dear :-(

MisteroUK, 2023-04-26:
tried many re starts but totally dead

MisteroUK, 2023-04-26:
ok PB14 on the y needed an ! in front so that's back but not working as should. Both the x and y are now both moving on a g28 but only one step

MisteroUK, 2023-04-26:
Strangely a g28 sends everything in the wrong direction ????? im getting confused now lol

bassamanator, 2023-04-26:
The homing sensitivity is too high, that's why it stops after a slight move.

Change both `driver_SGTHRS: 81`.

MisteroUK, 2023-04-26:
bingo!!!!!

ok we got a successful home. one question. when the home has finnished it sounds like the motors are locked and possibly under some form of resistance as they are getting hot and buzzing just idle. If I do an M84 this releases them and they go silent and cool down. Any ideas?

bassamanator, 2023-04-26:
Well they won't be *getting* hot, unless you've change the current. Do me a favour and paste a picture of the back of the x stepper, the amperage is listed there. In any case, the steppers are engaged because the printer has homed, and needs to know where the nozzle is.

Also please attach the `printer.cfg` to your next message, you might have to rename to .txt. Please also star the repo if you haven't already.

MisteroUK, 2023-04-26:
1.3a on the back of the motor

I will grab the printer config in a sec and send over

MisteroUK, 2023-04-26:
[printer.txt](https://github.com/bassamanator/Sovol-SV06-firmware/files/11337109/printer.txt)

[image]

MisteroUK, 2023-04-26:
The motors got REALLY hot while locked in doing a bed PID

bassamanator, 2023-04-26:
K same as mine and everyone else (except the first batch I believe).

Stepper motors are rated at ~125C. In any case, feel free to lower the current. Keep in mind, however, that the run_current is pulled from Sovol official marlin settings. 

☢ Having said all that, the motors should not be getting this hot, whatever that means. You might have a wiring issue here. Proceed with caution.

MisteroUK, 2023-04-26:
hmmm doesn't sound good :-( they were working absolutely fine and not getting hot at all with the original set up. Now they are roasting, really hot!!! doesn't sound safe at all with this set up, and I got so close.

bassamanator, 2023-04-26:
Turn off the printer now.

Are they all getting hot? Let me see the `printer.cfg`.

bassamanator, 2023-04-26:
Any which steppers are getting hot?

MisteroUK, 2023-04-26:
all of them

MisteroUK, 2023-04-26:
and they are whining

MisteroUK, 2023-04-26:
[printer.txt](https://github.com/bassamanator/Sovol-SV06-firmware/files/11337413/printer.txt)

bassamanator, 2023-04-26:
The printer.cfg looks fine, although I didn't see any changes to the `dir_pin`, are you sure this is the correct one?

As long as you feel that there are no wiring issues, you can do the following.
- Cut down `run_current` by half on all steppers that are getting hot.
- If they feel only warm to the touch, try printing something.
- If the print succeeds, you've good to go.
- If the print fails, reduce run_current further and print again.
- Repeat until you have a working printer and the steppers aren't getting crazy hot.

Hot is okay btw, sizzling hot is not ok.

MisteroUK, 2023-04-26:
I had the same idea to go down by half on all hot motors so I adjusted them to the following the attached config file below

[printer.txt](https://github.com/bassamanator/Sovol-SV06-firmware/files/11337569/printer.txt)

You will also see I have made some changes to the fan section as well. Let me know what you think :-)

Also I am currently using Prusaslicer and have some start and end GCODE im using. It worked fine with the original set up but wondered if you could have a look to see if it will be ok to use with this board and klipper set up?

*****START CODE*****

G90 ; Use absoulte coordinates 
M83 ; Extruder relative mode

M220 S100 ; reset feed rate to 100 percent
M221 S100 ; reset flow rate to 100 percent

M900 K0.07 ; Set Linear Advance K Factor

G28 ; Home all axes

M104 S165 ; set extruder temp pre-heat
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp

M420 S1 ; load abl mesh

M117 Start Position
G1 Z5 F150 ; lower nozzle
G1 X0.1 Y10 Z5.0 F5000.0 ; move to start position
G1 Z1.0 F150; Lower to ready

M104 S[first_layer_temperature] ; set extruder temp 
M109 S[first_layer_temperature] ; wait for extruder temp

M117 Printing Purge Lines
G1 Z0.26 F150 ; move lower
G4 S2 ; wait 2 seconds

G1 X0.1 Y150 Z0.3 F1500.0 E15 ; draw first line
G1 X0.4 Y150 Z0.3 F5000.0 ; move over a little
G1 X0.4 Y15 Z0.3 F1500.0 E30 ; draw second line
G4 S2 ; wait 02 seconds

M117 Print Start...
G1 Z0.6 F150 ; Lift A BIT
G92 E0.0 ; Reset Extruder
M83 ; Extruder relative mode
G1 Z1.0 F150 ; Move Z Axis up little to prevent scratching of Heat Bed

*****END CODE*****

; printing object ENDGCODE
G91 ; relative positioning
G1 E-2 F2700 ; retract
G1 E-2 Z0.2 F2400 ; retract and raise z G1 X5 Y5 F3000 ; wipe out
G90 ; absolute positioning

M117 Print Complete!
G27 P2 P ; park head & present print
M220 S100 ; reset feed rate to 100 percent
M221 S100 ; reset flow rate to 100 percent

M104 S0 ; kill extruder temperature
M140 S0 ; kill bed temperature
M107 ; kill fan

M84 X Y E ; disable all steppers but z

I may even take out the last bit of disable all steppers but z

MisteroUK, 2023-04-26:
so that z are also disabled

bassamanator, 2023-04-26:
That gcode will not work. Check the adjusting your slicer section for start end gcode.

Also do small test prints to get the current correct. Prints might take some time to fail, specially the heavier ones.

bassamanator, 2023-04-27:
The fan section looks fine, I'm sure it will make things quieter faster after the steppers stop.

Any successful prints?

MisteroUK, 2023-04-27:
So, I had a fail on the first print shown by the picture below:

[image]

So I put the stepper motor run currents back up. I have a benchy printing now and its going well and very nearly finished so I think the error could have been related to that, however the stepper motors are red hot so I am going to start to reduce the currents down and re print till I find a good working current.

bassamanator, 2023-04-27:
If possible, try to take a temperature reading of the steppers. I have what I would call a high pain tolerance, however, 65-70C stepper motors are painful to keep in contact with for more than 15 seconds, let's say.

I don't know if the plastic part holding the x stepper motor will survive a 'red hot motor'.

The error might be due to the mcu not being able to generate the required microsteps. If you encounter the error again at the default run_current, you can try reducing all microsteps to 64. Let's have a look at your `System Loads` section under the `Machine` tab.
[image]

MisteroUK, 2023-04-27:
this is one screen shot of the system idle......... I will wait for it to cool, set up another print and screen shot it while printing as well.

I also have a digital thermometer laser gun on order, should be here tomorrow so I can get some factual readings rather than just me feeling the motors.

[image]

bassamanator, 2023-04-27:
I would recommend reducing microsteps given that this mcu is a bit slower, won't affect much, if anything at all.

MisteroUK, 2023-04-27:
Here is a screen shot of the system loads while running a print. I haven't seen the MCU go above 12% at all and that is with 128 microsteps setting

One thing I will say is that the drivers on this board are the noisiest I've heard yet, sounds like an old pc game loading on tape lol.

Any adjustments to quieten them down at all do you know?

[image]

bassamanator, 2023-04-27:
The load has nothing to do with it.

At microsteps 64 or greater, things are as quiet as can be. You can turn on stealthchop to quiet things further, check klipper docs.

The klipper firmware, provided by BTT themselves, is quite old now from what I see in the picture, usually it's not an issue. Having said that, I would suggest you recompile it yourself according to docs found on the skr branch. It may help with your hot steppers.

MisteroUK, 2023-04-27:
Now that was my next thought so I think that is what I will be doing next :-)

I dropped the X anY to current values of 0.7 and the z at 0.8 and they are all at a reasonable temp for now.

inorichi, 2023-04-28:
I'm having the same issue, the steppers are whining until they are unlocked and they get really hot. I've been using the printer (the plus one) with the stock firmware for two weeks and they didn't ever make that noise. I don't know if they will degrade quickly like this.

I dumped the Marlin settings before switching and I can confirm the current values are correct on all steppers in your cfg so I think it should be something else (this is my first printer so I don't really know where the issue is).

I didn't make any change/mod to the printer itself btw.

MisteroUK, 2023-04-28:
Hi inorichi,

I think I may have solved this but have some more testing to fully confirm.

Try the following:

Enable stealthchop on the tmc section and adjust the current value to 0.580 on all 

should look something like below but remember im still testing so haven't confirmed the final best settings.

if you want to print fast not quiet then don't enable stealthchop.

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
run_current: 0.580
stealthchop_threshold: 0
interpolate: False
sense_resistor: 0.150
uart_address: 0
driver_SGTHRS: 81
diag_pin: PC0
stealthchop_threshold: 999999

MisteroUK, 2023-04-28:
Hey Bassam Husain,

Just to let you know I have carried on tuning the profile to get some results for you, I have attached below the printer config so you can have a look as your working on this project.

All in all I have this now printing successfully using prusaslicer using MK3s+ print settings as a base to start with. I have measured all temps of the stepper motors now based on the settings in the config file and all motors run between 35 - 45 degrees celsius.

I dropped the microsteps to 64 to give some head room and enabled stealthchop to quieten it down.

I compiled a new firmware and uploaded as well and now have it running fine with Klipper Screen.

All in all I have to thank you for help with my first steps over from Marlin into the world of Klipper :-)

If there is anything else you would like me to try or change to help then just give me a shout.

Printer Config File:
[printer.txt](https://github.com/bassamanator/Sovol-SV06-firmware/files/11355744/printer.txt)

System:
[image]

Pictures:

[image]

[image]

closed 2023-04-29 by MisteroUK

bassamanator, 2023-04-29:
@MisteroUK You're welcome.

Your machine is looking sharp. Love the KlipperScreen.

Please attach the newly compiled firmware, if possible.

Please star the repo, and watch for updates. Thanks!