bassamanator/Sovol-SV06-firmware #50: MECHANICAL_GANTRY_CALIBRATION not working properly

MirkoMorello, 2023-04-28:
Hi, I've been trying to use the MECHANICAL_GANTRY_CALIBRATION (G34 code) to calibrate my z axis, the problem is that the command doesn't seem to work correctly, the left motor doesn't push the left hinge of the rail all the way up.
The right part touches the top, after that, the left one pushes a bit towards the top but it doesn't have enough torque to push all the way through and align.
I tried rotating the coupler during the few seconds the steppers go off, but even though I do that, there still is a systematic error since doing another gantry calibration exactly after the first one shows the gap again, even if sometimes lower.
Thinking it was an HW problem, I tried reinstalling Marlin to see if the problem was reproducible with the stock firmware, and it doesn't look like it's the case because the gautry perfectly aligned, both hinges touched the top at the same time (after first calibration). This suggests that the gantry can't be pushed all the way up manually, it needs the motor to do this job correctly in order to perfectly align the two parts of the rack.
I tought I could bypass this problem with a SW solution like a simple bed mesh, but even with bed meshing for some reasons it still causes problems of not uniform distances from the bed while printing, leading to bad first layer adhesion in the right side of the bed.
I know that bad adhesion due to z axis misalignment is a common problem for SV06 users, and that the usual way to approach this is to do a manual calibration putting two identical pieces between the bed and the lower rail of the z axis rack and loosening/tightening the couplers until both of them touches slightly, and this seems to have fixed the problem (bed meshing suggests there is a delta of 0.2/0.3 mm), but it's a shame that I can't use the automatic gantry calibration with klipper since it seems's I'm one of the few lucky people that have a good calibration after a g34 in the stock firmware.
I guess that this is a problem of the MECHANICAL_GANTRY_CALIBRATION code present in the cfg folder, since it's been transposed from a Ender 3 printer implementation of the same feature (most likely) or that there is some tuning problems that led to different full revolution of the z motors (less likely).

bassamanator, 2023-04-29:
I've heavily tested and modified the macro for improved functionality. It works just fine.

Apparently, there have been 2 different kinds of motors that were shipped. *Your* steppers need more current. Even with the same steppers, it's possible for one user having to increase current compared to another user with the same steppers.

The default current set in the macro is `0.2`, increase it very slowly: `MECHANICAL_GANTRY_CALIBRATION CURRENT=0.25`. Be ready to do an emergency stop at a moment's notice.

Make sure you've pulled all the latest changes, and make sure your z couplers are secure.

MirkoMorello, 2023-05-01:
Hi Bassam, to my great pleasure you were right!
Now it aligns perfectly, but it has a little bit of coil whine, I'll try with lower current. By the way, does increasing the current as I did lead to problems on the long run? I personally modified line 69 of the function, putting `SET_TMC_CURRENT STEPPER=stepper_z CURRENT=0.25 ; drop current on Z stepper`, so the current gets from 1.0 to 2.5 (instead of 2.0) for the operation of the gantry calibration, I've tried to understand what `oldhold` is but `hold_current` is not present in the config file.
Unluckily, I found out that the alignment leads to worse calibration than I originally thought, after aligning, the bed mesh has a delta of 0.7mm, which is too high. I don't know why but stock marlin firmware did a better job with bed meshing, must be a little bit more aggressive with the corrections than the one in Klipper.
I'll try to make a shim roughly 0.5/0.7 mm to make the automatic alignment more precise, as of right now I'm stuck with manual calibration, which brings me to a delta of 0.2/0.3 mm, which is more due to normal bed deformation, I'll attach images about the results of manual calibration below.
Last question, how are you doing calibration on your printer? Do you have any suggestions?
Just wanted to say thank you for your work and for being so helpful to everybody, I appreciate it.

[image]
[image]

bassamanator, 2023-05-02:
You're welcome.

Lowering or increasing current during this process is a non-issue. Keep in mind, we're talking 0.0 to 1.0, anything higher could lead to blowing up your motors. You said 2.5 up there...

What was the range with marlin?

Shimming can work, the silicone mod is another option. This model is notorious for x-twist as well.

Try to printing a layer test to see if there really is a problem. Use a filament where the extrusion multiplier is 1.
[parametric_bed_mesh_test_generator-30mm-SovolSV06-extrusion_multiplier1.tar.gz](https://github.com/bassamanator/Sovol-SV06-firmware/files/11369019/parametric_bed_mesh_test_generator-30mm-SovolSV06-extrusion_multiplier1.tar.gz)

MirkoMorello, 2023-05-02:
Apologies, I edited my previous message, I meant to say 0.25
With Marlin I couldn't check the range because you don't have this information, I just saw it had a better first adhesion for a 10/15cm object placed in the center right after the gautry automatic calibration, but I'm starting to blame myself due to poor testing, I should have tried to print something larger like the one you just shared. It definitely looked better than the one with klipper after automatic calibration but I'm lacking information to assess it was perfect, which I honestly don't think it was.
I'm starting to see problem with adhesion at the extreme edges of my prints so I think it could be caused by the x-twist as well, any suggestions for fixing it?

I'll try your test and post about it in a few days, I'm not home and can't test it right now.
I'm new to 3D printing so I've never tested the extrusion multiplier of my filaments, I'll do it as accurate as I can and see if I have one with multiplier to 1.
In case it's not 1, if you can share the stl I'll change the multiplier to the one I'll find and slice it.

closed 2023-05-19 by bassamanator

iJebus, 2023-05-28:
Not to revive a dead topic, but I feel like my question is related. Running G34 on my SV06 fails, with only one stepper moving for the bump. With steppers off and turning them manually, I notice the stepper not moving is very stiff or crunchy, for lack of a better term. Does that indicate a bigger issue or would you recommend bumping up the current as above, until they both move?

bassamanator, 2023-05-28:
Always good to revive an old thread.

>  With steppers off and turning them manually, I notice the stepper not moving is very stiff or crunchy, for lack of a better term.

Are you saying that the z steppers are always 'unsmooth'? That sounds like the bearings need lubrication. See Savage Lau's video on this. Putting grease on the rods provides the most minor form of lubrication btw.

Having said that, it seems that you're in the same boat as the OP. Follow my instructions to him, always being ready to hit emergency stop during gantry calibration.

iJebus, 2023-05-29:
Haha, so, TLDR, bumping the current to `0.45` worked and gantry ramming works. 🚀

What I was trying to explain is that I noticed while the printer was powered off, manually turning one of the steppers was more difficult. I wondered whether that might indicate some sort of pending mechanical failure. That's getting pretty far outside the scope of this issue though. Thanks for replying and thanks for everything you've put together in this repo!

bassamanator, 2023-05-29:
Great!

For optimal performance, both sides of the motion system should work the same. Check to make sure the z couplers are well tightened though don't over do it. Maybe also check bolts holding the z steppers. I don't think it's a big issue, but something you might want to look into when you have time. Best!