bassamanator/Sovol-SV06-firmware #54: Screen on SV06 gets messed up on longer prints

Odinios, 2023-05-15:
On longer prints >3 hours, sometimes the screen on the printer bugs out and symbols change / not all info shows up correctly. WebUI  (mainsail) still seems to function properly.

When I was running a webcam the issue was more common and froze the WebUI and eventually the print stopped. Currently not using any webcam.

A firmware restart will fix the issue, just can't do that mid print, also don't want the print to randomly stop...

I have tried changing the USB cable from Raspberry Pi to SV06 and reseated the screen cable.

Below is an image when it messes up. *Does not always look like this, sometimes it's different:
[image]

If this isn't the right place to post this or if it needs to be removed, let me know (new to GitHub)

bassamanator, 2023-05-15:
This isn't an issue related to the OSS config, but let's try to solve. The screen on this printer is somewhat problematic, I've seen countless posts about it dying, or bugging out worse than what you've posted, so it is probably just that.

What are the specs of the RPi?

Can you also do the mod found [here](https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574)? I'm not saying this will solve your issue, but the RPi is providing *some* power to the printer, which is dangerous regardless, and I wonder if this is potentially causing an issue.

Odinios, 2023-05-15:
Thanks, I'll try that post once the print finishes. It's raspberry  pi 4b 4gb model.

Odinios, 2023-05-15:
I'm gunna give the print another try now, will report back in a few hours. I suspect the pin mod on the cable will solve the issue. Thanks again!

closed 2023-05-16 by Odinios

bassamanator, 2023-05-16:
@Odinios So did that work?

Odinios, 2023-05-16:
@bassamanator I ran the print again (with the fix) and it didn't mess up! So I'm assuming it's all fixed

bassamanator, 2023-05-16:
@Odinios Great! 😀

Please star the repo if you found it useful.