bassamanator/Sovol-SV06-firmware #55: [BUG] Z offset is not saved/applied

bladyle, 2023-05-19:
**Describe the bug**
I have installed klipper on old laptop and everything work except I cant see that my z offset is applied when I'm printing. 
I have terrible first layer and bed adhesion. 
I have done the step for z offset calibration from first page and typed save_config after that. I can see my value on the bottom of the printer.cfg but no on the z offset widget on the dashboard. 

**What kind of device are you using to host Klipper?**
 - OS: Linux Mint on Laptop
 - Version: latest
 
**How did you install Klipper?**
 -KIAUH,

**Additional context**
Add any other context about the problem here.

Value from config file is: 
#*# [probe]
#*# z_offset = 1.505
but when I set it up it is negative if that's metter.

bassamanator, 2023-05-19:
You don't have to worry about applying the z offset manually somewhere, as long as you've done a `probe_calibrate`, `save_config`, you're good to go.

The widget displays any babystepping z offset values, it doesn't display your `z_offset` from the `printer.cfg`.

bassamanator, 2023-05-19:
This is not an issue. Please create a Discussion about your bed levelling issues. Unfortunately, you probably suffer from x twist; please see some info about it in the discussions.

bladyle, 2023-05-19:
Yes I have done probe_calibrate and save_config but still I can see my first layer is not correct. I may have x twist and my bed is not perfect but I have some pretty good prints with stock firmware before flashing klipper (tried with different configs also) and if i exclude this issue I still have good prints but I'm pretty sure z offset is not applied on my prints .. 
Is it a problem that I compiled .bin file myself and not flashed the one from your files?

bassamanator, 2023-05-19:
Compiling your own firmware is fine.

This doesn't look quite right to me:
```
## [probe]
## z_offset = 1.505
```
It should look like this (from my config):
```
#*# [probe]
#*# z_offset = 1.610
```
Do you have such a section in your `#*# <---------------------- SAVE_CONFIG ---------------------->` secion in your `printer.cfg`?

bladyle, 2023-05-19:
Yes i have that section and my values are exactly like yours with #*# at the front only the value is different.

bassamanator, 2023-05-19:
If that is the case, then `z_offset` is being applied, it doesn't require a manual load.

Did you see the `adjust your slicer` section?

bladyle, 2023-05-19:
Yes I'm using suggested slicer - super slicer with the bundle profile and The relevant macros for print start and end.

bassamanator, 2023-05-19:
That means your mesh is being loaded. While printing the first layer, depending on your mesh, you should see the screws rotating in some areas.

Closing as this is not an issue related to the config.

Feel free to create a discussion about this.

closed 2023-05-19 by bassamanator