bassamanator/Sovol-SV06-firmware #62: Thoughts on enabling the SV06 Plus LCD?

ccatlett1984, 2023-06-09:
It has been mentioned on the Sovol support forum that it might be possible to enable the stock LCD screen for use with klipper.

bassamanator, 2023-06-09:
Yeah it's something that I want to do. Are you volunteering yourself as a tester/helper?

ccatlett1984, 2023-06-09:
I'm currently getting a Pi together and will be getting klipper installed shortly. I'm not a software dev, but have been doing IT consulting for 10+ years now, and would be happy to assist with testing/feedback.

NyoX01, 2023-06-10:
In my SV06 Plus FB group someone sent a link to a repo that enables the stock screen.
I'm completely new to Klipper and was asking about setting it up in the group, someone then put up this link.  I will post it here and maybe someone that has way more knowledge than I can try it out, or I will once I get enough nerve to actually try to klipperize my Plus.  OMG I'm so scared I'm going to screw something up...
https://github.com/fryc88/klipper-sv06plus-screen?fbclid=IwAR0CJ0Gw6iIG820jVakwPea6_GI7Lw-o1nKy6PZJLxZyB4XdeXWpSsdV5-0
If this is not allowed, I'm sorry, just trying to help someone that may want that screen up and running.

bassamanator, 2023-06-10:
That is the work that I was going to pull into the `sv06-plus` branch. I find it could be a bit more user friendly. My concern at this point is that the klipper repo this work points to will never be integrated into klipper official, which is not necessarily a big deal.

In any case, this has been on my todo list for well over a month and remains as such at this point. Hope to get to it eventually.

NyoX01, 2023-06-10:
So Bassamanator, you are a legend! I thank you (like so many before me), for putting this together. But is there a fairly easy straightforward way to klipperize the Plus? Like an idiot's way lol? I've never really messed with Linux or Klipper and I even went as far as to download the write-up Rory had in our group. But after reading it several times there is terminology that I'm not familiar with. So once again, I'm terrified I'm going to screw something up...

bassamanator, 2023-06-10:
@NyoX01 Well, I would love to tell you that my instructions are idiotproof, but I know from experience that an idiot will always prove you wrong 🤣.

I want you to go read the full [documentation](https://github.com/bassamanator/Sovol-SV06-firmware/tree/sv06-plus) on the sv06-plus branch, let me know when you're done, and then I'll say more. It's about a 6-minute read.

NyoX01, 2023-06-10:
I went over the instructions several times, and printed them out so I could reference them as needed :)  I also sent you a message so we don't take up precious space on here.

bassamanator, 2023-06-10:
@NyoX01 Don't worry about space, and great!

So now that you've read the docs, I can tell you it's very difficult to 'brick' these boards (render them useless). If you start on this klipper path, and decide it's not for you half way through, you can always flash marlin back and be good to go again.

Just follow the instructions step by step, don't skip anything, and you'll be fine. Feel free to create a discussion if there are issues, I've walked many users through this process.

RandomSpaceship, 2023-06-12:
I've gone through the mods, and got the screen working. Most features work, but as noted in the repo, mesh leveling doesn't - in fact, it and all the other level commands crash Klipper!
If I have time, I might see if I can fix it, but no guarantees.

RandomSpaceship, 2023-06-12:
It works, but attempting to home the printer from the touchscreen doesn't. Once it starts an X/Y home the Klipper service on the Pi immediately crashes. Mesh leveling also immediately crashes, possibly due to attempting to home first. Outside of that, everything works as expected. Homing from the Mainsail interface or uploaded g-code works fine.

bassamanator, 2023-06-13:
@RandomSpaceship Interesting. I'm glad I haven't invested time in pulling in those changes. Curious, have you pointed mainsail to the [gbkwiatt](https://github.com/gbkwiatt/klipper) repo?

RandomSpaceship, 2023-06-13:
Correction: Klipper's not crashing, the printer firmware is. I didn't read the error properly.

Anyway, since I discovered that, I uploaded the `firmware5.bin` file from the repo (I built it myself before). It now works perfectly after a restart - even mesh bed leveling. There are still graphical issues (text boxes not properly aligned), but I can live with that. I haven't tried mechanical Z align since it's not great and I've aligned it manually.

ChrSch84, 2023-06-14:
@bassamanator Hello is there a "how to" or so to test it with the display. Your Klipper Konfig is working here @ 06 Plus but at the Moment without display.

Thanks for the hard work.

RandomSpaceship, 2023-06-14:
Yet Another Update: Homing crashes with the `firmware5.bin`! It seems inconsistent as to when it does or doesn't home. I don't have the time right now to run more tests and work out what exactly is happening, but it's worth noting.

bassamanator, 2023-06-14:
@ChrSch84 Well these are 2 separate entities, this repo and `fryc88's` guide to get the screen working. If you already have the OSSKC working on your Plus, you could go through the guide to try and get the screen working. Those encountering issues, @RandomSpaceship, should report the issues on those 2 repos, and hopefully get them fixed.

ccatlett1984, 2023-06-18:
I have the screen enabled, and most things seem to be working on my Plus.

bassamanator, 2023-06-18:
@ccatlett1984 Glad to hear it. I'm going to keep this issue open. Will call on you if/when I work on this.

Quick question: you are point to the [gbkwiatt](https://github.com/gbkwiatt/klipper) branch to make all this work?

ccatlett1984, 2023-06-19:
Correct, I set the gbkwiatt repo in KIAUH and installed everything through there.

fryc88, 2023-06-23:
> It works, but attempting to home the printer from the touchscreen doesn't. Once it starts an X/Y home the Klipper service on the Pi immediately crashes. Mesh leveling also immediately crashes, [...]

There's something to do with senseless homing & calling home for x y and z separately via the screen, I haven't had time to go that deep with it, but to be fair, who's doing mesh or homing via screen?
I had this up and running for ~6-7 months now on my SV01pro and homing issues shows up less often there, I'd say occasionally.
And no, it's not immediately crashing, it stops when it reaches the point of virtual end stops for some weird reason...
I'm using this only for basic stuff like adjusting offset, temps, etc. while printing, also, I'm happy anyway about having possibility to do so, instead of having a blank screen. Because of it's construction it will NEVER work as a proper klipperscreen, and original author of it (which is not gbkwiat, gbkwiat helped a lot to get it working on sovol screen and he set up that branch to auto merge with main klipper branch) has abandoned it completely.
Recently we've managed to sort max heating temps in the screen, because originally it was -15 degrees from temps set in the printers firmware, just FYI ;)

closed 2023-07-30 by bassamanator