bassamanator/Sovol-SV06-firmware #65: Q: How to config for cooler stepper motors?

bladyle, 2023-06-14:
Hi, I'm running your config file without a problem but I have noticed that my stepper motors run hotter that stock firmware. What is the proper way to config them even if I have to sacrifice little bit of performance.

RandomSpaceship, 2023-06-14:
Run lower currents. In `printer.cfg`, under `[tmc2209 stepper_AXIS]`, there's a `run_current` variable. This is measured in amps delivered to the motors. Dropping it will reduce how much power gets sent to the motors, and therefore how much power is heating them.

However, IIRC the base Klipper config is the same as stock. If you're using print profiles with higher acceleration, that means that the motors need to make more force to move the axes around. As such, they might be getting _closer_ to their maximum current (they might not have been running at it before).

So, to keep them cool, you have two options: Reduce acceleration and reduce the current. However, since there isn't any fancy electronics in them, you can actually run them to be uncomfortably hot. If you can't touch them at all, that's too hot, but they can get surprisingly warm safely.

bladyle, 2023-06-14:
Yes I have just found my screenshots from stock firmware and compared the current there, just to confirm it's the same as you have wrote. I have occasionally run higher acceleration prints but not always and they are hotter even with slower prints. I'm sure they are still in  safe zone but want to try to reduce temp if it's possible. Do you know if reducing the microstep value will help lower the temp ?

RandomSpaceship, 2023-06-14:
I'm not sure, it's not something that I've looked into. I think it's exposed in the advanced settings menu in the stock firmware if you want to load that and compare it. Have you loaded the same g-code (or as close as you can get - same jerk/acceleration/speed settings) in both the stock and Klipper firmwares, and measured the motor temps? I find it very odd that the same print settings and motor configuration settings would make the motors hotter. However, if you still think they're running hot after whatever changes you make, motor heatsinks aren't very expensive on Aliexpress.

bassamanator, 2023-06-14:
@bladyle These motors can get as hot as 125C and be fine. They shouldn't be getting anywhere near that, however. Reducing microsteps shouldn't play a role here, but you could try it I suppose: 16, 32, 64, 128, and even 256 are valid values. It would help if you could take actual temperature readings, motors shouldn't be hitting the 60C mark.

In any case, feel free to keep reducing `run_current` until you start skipping steps.

closed 2023-06-17 by bassamanator