bassamanator/Sovol-SV06-firmware #70: Fan controll of the controllboard fan

Remleyfpv, 2023-06-18:
I dont know if it is possible but would be cool if the controll board fan is controlled by the MCU temp of the STM. But I cant find what PIN this fan would be. The pin header is right next to a controllable fan(part cooling fan) so that leaves me to belive that it is a controllable fan output not just a 24V port.

bassamanator, 2023-06-18:
>The pin header is right next to a controllable fan(part cooling fan) so that leaves me to belive that it is a controllable fan output not just a 24V port.

Unfortunately, that is not the case. Not possible via software/config from what I've seen, and I've looked into this quite a bit.

Do contribute by creating a PR if you find a solution.

closed 2023-06-18 by bassamanator

Remleyfpv, 2023-06-18:
oke and just to make sure: is yours turned of always to? or is mine just broken?

bassamanator, 2023-06-18:
Everyone's hotend cooling fan is always turned on for the SV06/Plus. It's not a controllable fan.

bassamanator, 2023-06-18:
Sorry, the motherboard fan turns on when the TMC drivers come into play, I assume. I say 'I assume' because I haven't found the pin config for it, nor does it listed in the official Marlin source code. What I do know that the motherboard fan turns on *only* when I start printing.

bassamanator, 2023-06-18:
[Here's](https://forum-drucktipps3d-de.translate.goog/forum/board/42-sovol/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp) a forum that contains a lot of information, however, I have not found any of controlling 2 or the 3 fans, only the part cooling fan is controllable. It's in German but that link is running through google translate.