bassamanator/Sovol-SV06-firmware #76: Draw Purge Line[BUG]

Chewyyy, 2023-07-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a Print
2. Print doesn't draw purge line
3. If you try and run draw_purge_line alone, it gives error "Error evaluating 'gcode_macro draw_purge_line:gcode': ZeroDivisionError: float division by zero"

**Expected behavior**
Drawing of a purge line before prints begin

**Screenshots**
If applicable, add screenshots to help explain your problem.

**What kind of device are you using to host Klipper?**
 - RPi 4B, up to date as of 07/22/2023

**How did you install Klipper?**
 - Mainsail

**Additional context**
No other macros being run other than your package

bassamanator, 2023-07-22:
There is no draw_purge_line macro included in this config. Please read about  purge_line macro in the docs.

closed 2023-07-22 by bassamanator

Chewyyy, 2023-07-23:
You are absolutely right and I am an idiot.  Sorry!!!