bassamanator/Sovol-SV06-firmware #77: Questions Related to this Repo / Unknown Commands

Chewyyy, 2023-07-23:
Hi!

I just did a clean install of Mainsail OS and followed only your instructions and am using your macros / firmware setup.  I have also gone into Cura and setup the Gcode before/after prints as recommended.  I have done all the calibrations and have everything working, but just have a few "issues".

During prints, my console window is being spammed with "Unknown command" errors for M205 and AFTER_LAYER_CHANGE and BEFORE_LAYER_CHANGE.  Additionally, when I initially start a print, I get the same Unknown command error, but for "PRINT_START_SET"and "PRINT_START_SET".  I've attached the logs.

I appreciate any and all assistance! Thanks in advance!
[klippy (3).log](https://github.com/bassamanator/Sovol-SV06-firmware/files/12138047/klippy.3.log)

Chewyyy, 2023-07-23:
M205 issue was the acceleration controls got turned on in Cura.  I'm assuming the Print_start_set and before/after layer change is a similar setting.  If you happen to know which setting, it'd be greatly appreciated!

Chewyyy, 2023-07-23:
It ended up being macros that were nested into Cura.  Nothing to do with your wonderful firmware set! Sorry for wasting your time!

closed 2023-07-23 by Chewyyy

bassamanator, 2023-07-24:
@Chewyyy No worries! 😁