bassamanator/Sovol-SV06-firmware #83: [BUG] Screws_Tilt_Adjust coordinates to nozzle instead of probe

race85com, 2023-08-03:
Current values in Screws_Tilt_Adjust reffer to position of screws to nozzle coordinates - not probe to screws coordinates. As a result, probing during SCREWS_TILT_CALCULATE is moved by offset of the probe, resulting in incorrect calculation.

I would suggest to change the config.cfg to these values, so that probe hits screws coordinates during SCREWS_TILT_CALCULATE

[screws_tilt_adjust]
screw1: 84.5, 136.5
screw1_name: center
screw2: 0, 51
screw2_name: front left
screw3: 170, 51
screw3_name: front right
screw4: 170, 221
screw4_name: back right
screw5: 0, 221
screw5_name: back left
speed: 60
horizontal_move_z: 10
screw_thread: CCW-M4

bassamanator, 2023-08-03:
This is pulled from [master](https://github.com/bassamanator/Sovol-SV06-firmware/blob/fea7a35c3af5aae0470c55aa9b0f2c119a4276c1/printer.cfg#L195C11-L195C11):
```
[screws_tilt_adjust]
screw1_name: center
screw1: 84.50, 135
screw2_name: front left
screw2: 0, 51
screw3_name: front right
screw3: 169.75, 51
screw4_name: back right
screw4: 169.75, 221
screw5_name: back left
screw5: 0, 221
horizontal_move_z: 10
screw_thread: CCW-M4
```

Our two `screws_tilt_adjust` sections deviate by 1-2mm from what I see. Perfectly normal to have such a variance between the same model of printers at this price range.

Am I missing something here?

race85com, 2023-08-03:
I did not have the updated version of the master, now I see you have done a correction in the config a month ago, sorry for the confusion

bassamanator, 2023-08-03:
No worries my friend.

There's a pinned discussion explaining the update process. I suspect that you're a git user so it will be straight forward for you. 

Feel free to open other issues! Best!

closed 2023-08-03 by bassamanator