bassamanator/Sovol-SV06-firmware #85: Possible optimization on bed heating when close to target temperature?

Nukle0n, 2023-08-05:
Maybe this is a general Klipper thing but I figured I'd ask here since i'm using your config (thank you for making these by the way)

Sometimes when I've cancelled a print to quickly change something an then started it again, the temperature of the build plate has only fallen very little, like under 1 degrees celcius. But it still takes ages for it to get back up to the full 60 or 80, as if the printer is extremely scared of going slightly over, so it only heats extremely little, this means it will take 5 minutes for the printer to go from 59.4 to 60c. I've done PID tuning again to try and see why this could be, but it doesn't seem to help, it seems maybe the automatic parameters for calculating how to heat up the bed aren't optimal for when already close to the target temp?

bassamanator, 2023-08-05:
I've had a similar experience, and it's quite annoying. This is related to Klipper, not to this config. 

Replace the following in your `cfgs/misc-macros.cfg`:
```
[gcode_macro M190]
rename_existing: M99190
gcode:
    #Parameters
    {% set s = params.S|float %}

    M140 {% for p in params %}{'%s%s' % (p, params[p])}{% endfor %}   ; Set bed temp
    {% if s != 0 %}
        TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={s-1} MAXIMUM={s+1}  ; Wait for bed temp (within 1 degree)
    {% endif %}
```

The `s-1` in `MINIMUM={s-1}` is saying that the minimum temp can be *under by 1 degree* of the *requested* temperature. Caveat, this will apply to all bed temperature settings, ie, every time you set a bed temperature.

Another caveat, this is just a 'hack', it is an attempt at compensating for a larger issue. Please create an issue here https://github.com/Klipper3d/klipper/issues.

bassamanator, 2023-08-05:
Also please let me know if this worked.

Nukle0n, 2023-08-05:
Thank you for the quick response, I'll try and put it into my config, not sure when it'll happen next but I'll try and remember to get back to you.

Sadly it appears you cannot make a github issue for Klipper, the "new issue" link gives you 3 options which are respectively reading their documentation, posting on their forum, or joining their Discord server.

bassamanator, 2023-08-05:
Sounds good. I'll try to talk to klipper guys/post on their forum.

Feel free to re-open issue!

closed 2023-08-05 by bassamanator

Nukle0n, 2023-08-07:
user "why_me" recommended 
```
[gcode_macro M190]
variable_temp_window: 5
variable_temp_boost: 10
rename_existing: M99190
gcode:
  #Parameters
  {% set set_temp = params.S|float %}
  {% set current_temp = printer.heater_bed.temperature %}
  {% set boost_temp = [set_temp + temp_boost, printer.configfile.settings.heater_bed.max_temp]|min %}
  {% set diff = set_temp - current_temp %}
  # safety
  {% if set_temp > printer.configfile.settings.heater_bed.max_temp %} { action_raise_error("ERROR: Bed temperature was set above bed limits") }{% endif %}

  {% if diff > 0 and diff < temp_window %} # only when heating up and close to target temperature
    M140 S{boost_temp}
    {% set temp_max = boost_temp %}
  {% else %}
    M140 S{set_temp}
    {% set temp_max = set_temp %}
  {% endif %}
  {% if set_temp != 0 %}
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={set_temp-1} MAXIMUM={temp_max + 5}  ; Wait for bed temp (within 1 degree)
  {% endif %}
  M140 S{set_temp}
``` 

Which seems to work decently,  making it start 1 degree short. with the KAMP bed levelling it evens out to the target for the most part anyway.