bassamanator/Sovol-SV06-firmware #91: [BUG] Klipper crashes/turns off after setting hotend temperature after M600/filament change

RedTheKitsune, 2023-09-05:
**Describe the bug**
On Sovol SV06 Plus, printer turns off after setting hotend temperature after M600/filament change gcode.

**To Reproduce**
Steps to reproduce the behavior:
1. Add M600 gcode on any layer
2. Wait for printer to execute it (move printhead forward)
3. Hotend temperature will be set to 0C
4. Set hotend temperature to anything
5. After like 5s printer will shutdown

**Expected behavior**
Not reset hotend temperature after M600/filament change gcode or fix the crash caused by setting hotend temperature.

**What kind of device are you using to host Klipper?**
 - OS: Odroid XU4

**How did you install Klipper?**
 - KIAUH

bassamanator, 2023-09-05:
This has nothing to do with the OSS config because I use filament change all the time. And the SV06 (me) and the Plus (you) share the same PCBs, so our experiences should be the same. The difference here is our operating systems, therefore this is an Odroid XU4 issue.

I'll have a quick look at the logs for you before I close this issue though, attach them here.

Also, are you up to date with this repo?

RedTheKitsune, 2023-09-06:
Soo I think this was caused by faulty USB cable or SBC USB port.
Found it because after updating config to the latest one, printer got randomly disconnected during homing procedure, so 100% it was an hardware issue.

closed 2023-09-06 by RedTheKitsune