bassamanator/Sovol-SV06-firmware #92: [BUG] Absolute / Relative mode

Regulate3772, 2023-09-10:
**Describe the bug**
I am not sure if I'm the only one with this problem but after quite a long research I want to share my path.

In the macros it seems that the extruder is supposed to work in absolute mode but in the linked slicer settings it is set to relative mode. This caused my prints to fail.

In the macro file, you define in which mode the extruder moves are calculated. You used     
M82                                     ; Absolute extrusion mode

Yet in the slicer profiles (both superslicer and prusa), they use relative mode under: Printer Settings - General - Advanced - Use relative E - distances.
When working with your macros, this box needs to be unchecked or this error shows up in mainsail which cancels the print:  
"Move exceeds maximum extrusion (4.357mm^2 vs 0.640mm^2)"

There is an explanation about absolute / relative mode here:
https://klipper.discourse.group/t/problem-move-exceeds-maximum-extrusion/3941/6

So either the macros needs to be changed to relative mode, or there should be a comment when linking the slicer profiles that the box needs to be unchecked to match the macros.

If I'm wrong or messed up my settings myself, I'm sorry. 

**How did you install Klipper?**
 - KIAUH

bassamanator, 2023-09-10:
Thanks for creating this issue! I'll try to reproduce it on my end with a fresh OSS install.

A few quick questions:
- your printer model?
- when did you download this configuration? I just want to make sure that your local copy is up to date with this repo.

Regulate3772, 2023-09-10:
My model is Sovol SV06 Plus and I installed klipper (and your configs) about a month ago.

The print failiure comes only in some prints, it's more about the interaction of the slicer config and the macro file. It works perfectly fine when using the M82 in the macro as it is (absolute) instead of M83 (relative) if the described box is unchecked in the slicer.

I also know how it came to the mismatch.
Looking at Ellis changelog at https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles
you can find:

> 2022-04-14:
> use_relative_e_distances back to enabled.
> Some people had M83 in their PRINT_START macros which didn't agree with this.

This means, about a year ago he changed the profile to relative mode while in your macro you use M82 (absolute) which causes errors at some prints.
In my opinion absolute is the better option. Yet, maybe you want to include in your description that you have to uncheck the box in prusa / superslicer for it to work with the macros.

bassamanator, 2023-09-11:
Having trouble reproducing.

Are you able to 'crash on demand'?

When you send a print like this:
[image]
it crashes?

Then when you just uncheck:
[image]
it prints just fine?

Regulate3772, 2023-09-11:
Exactly, both in prusaslicer and superslicer. It happens on very tall prints since the flow is calculated is not correct. Mostly on the second layer. 

Besides reproducing: it is still fully possible that I messed up another setting although I reinstalled, downloaded and tested again. But other than that: it seems like you did program the start print code in absolute mode, right? You even commented it the macro file that it uses absolutes. So the slicer settings have to be matched to the macro file, doesn't it? Just in a programming way. The box in the slicer is exactly for that: to match it the start print, pause, filament change ect. (All in absolutes in the macro file)

bassamanator, 2023-09-13:
Unable to reproduce. Added note about this to discussion#13. Thanks.

closed 2023-09-13 by bassamanator