bassamanator/Sovol-SV06-firmware #105: [BUG] Firmware version included in misc folder is not up to date and can lead to problems (e.g. mcu_temp readout crashing)

Errandir-of-Gondolin, 2023-11-05:
I've recently installed Klipper on the SV06 (using KIAUH and the provided cfg files and firmware in this repository - Klipper install is on an OG RPi4). But I ran into the issue that when I added the mcu_temp readout in the cfg file, it would read as -409°C in Fluidd. That is way outside the limits (including those of the physical world), causing the printer to shut down. My other 2 printers (not SV06), that run from the same RPi4, did not have this issue.

Rebooting Klipper and/or the printer didn't change anything. Only commenting out the mcu_temp section again would allow me to actually control the printer. The code I used to activate the mcu readout was:

```
[temperature_sensor mcu_temp]
sensor_type: temperature_mcu
sensor_mcu: mcu
min_temp: 0
max_temp: 100 
```

Klipper, Moonraker and all OS packages were updated to the latest versions (on 05/11/2023). Based on a hunch from a friend, I manually compiled the firmware again via Klipper, flashed that, and it solved the problem. So any users using the 'old' firmware in the misc folder might see the same issues pop up if they add the mcu_temp section to the cfg file when using newer versions of Klipper.

I would suggest either removing the firmware file, or providing an update of the firmware, compatible with the latest version of Klipper.

Image below shows the readouts in Fluidd. The jumps to 0°C are at the timestamps of firmware restarts.
[image]

After compiling new firmware, using the settings below, and flashing it to the printer, all worked as normal again.
[image]

Output after flashing new firmware: 
[image]

bassamanator, 2023-11-05:
I've known about this issue since the early days of this printer. Glad it's fixed! Will compile and test.

Thanks for reporting.

Errandir-of-Gondolin, 2023-11-06:
Another day, some more info. Sadly what I previously reported might not be an actual fix.

Context: today I resumed my calibration of the SV06. I started with the paper test, to set the z-offset. As soon as my nozzle dipped below 0 height (at which point the nozzle still didn't touch the bed), I got the same error: MCU temp out of range, back to -400°C. I recalled it was at the same exact moment in my calibration yesterday that this first popped up. So that got me thinking.

Sure enough, even reflashing the latest firmware didn't do anything. Probably because the probe was still at the same distance from the bed when rebooting the firmware, causing the same issue.

Temporary solution:
- commenting out the MCU_temp section of the printer.cfg file
- restarting firmware
- moving nozzle up
- uncommenting MCU_temp section
- restarting firmware ==> no more error

But as soon as you lower the printhead low enough (but still before the nozzle is close enough to the bed), the same error occurs and you need to start the process all over again.

I dialled in the z-offset via probe_calibrate, without the MCU_temp section in the cfg file, and I managed to set it to a value (default was 0, calibrated was around 1.5). I hoped that with a properly calibrated z-offset, the probe wouldn't be going too low for this to happen anymore, but sadly it is very repeatable. I don't even have to go to negative z-height values; lowering the nozzle to 1mm off the bed is enough for the error to trigger and shut everything down. The probe still is about 2.5mm above the bed at that point.

I have no idea how a seemingly probe-related issue can lead to unrealistic mcu temperature measurements, but here we are. For now I need to leave the mcu_temp section out of the config file to be able to proceed, but I'm concerned about the mcu not being monitored (including for too high temps). Not sure if there is actually any protection built into Klipper based on mcu temps going too high, but if there is, it might not function and hence increase the fire hazard of the printer.

bassamanator, 2023-11-06:
Ah what a pity. Thanks for the hard work looking into this. Please consider pasting links to where you asked about this and what info you found, in case someone wants to look into this in the future.

closed 2023-11-06 by bassamanator

Gronal, 2026-09-12:
## Problem

Self-compiled Klipper MCU firmware never successfully flashes via the
microSD bootloader on my SV06 mainboard (`SV_NL V1.3.1`, GD32F103RET6, CH340
USB bridge). The MCU keeps running the original precompiled
`klipper-v0.13.0-371-g7a723bdc.bin` from this repo's `misc/` folder, no
matter which `menuconfig` combination I try.

I originally hit this because of issue #105 (`mcu_temp` reads
`-409.884`, causing an ADC-out-of-range shutdown) — my MCU firmware is
older than my Klipper host (`v0.13.0-745-gf0892d82`), which matches
exactly what @Errandir-of-Gondolin described there. Their fix was to
self-compile, so I've been trying to do the same, without success.

## Hardware

- Board: `SV_NL V1.3.1`
- MCU: GigaDevice GD32F103RET6 (STM32F103-compatible clone)
- USB bridge: CH340 (`1a86:7523`, kernel driver `ch341`)
- Host: Ubuntu 22.04.5 LTS on an x86 laptop (not a Pi)
- Klipper host version: `v0.13.0-745-gf0892d82`
- MCU currently running: `v0.13.0-371-g7a723bdc` (precompiled, from
  `misc/`)

## What works

- The precompiled `klipper-v0.13.0-371-g7a723bdc.bin` from `misc/`
  flashes and runs fine via the microSD bootloader.
- I've confirmed the flash **procedure** itself works: USB cable fully
  unplugged during flash, SD card FAT32 with 4096-byte allocation unit,
  file renamed to exactly `klipper.bin`, `sync` before `umount`, 90+
  seconds powered on before removing the card.

## What I've tried (self-compiled firmware never takes effect)

Confirmed via `md5sum` that the file on the SD card matches
`out/klipper.bin` exactly, and the file size is reasonable (~40 KB,
well under any flash limit). After every attempt below, the MCU still
reports `v0.13.0-371-g7a723bdc` post-flash — i.e. the bootloader
appears to never actually accept/write the new file.

**menuconfig combinations tried (all other options identical: STM32 /
STM32F103, 8 MHz crystal, Serial USART1 PA10/PA9, baud 250000):**

| Bootloader offset | Disable SWD | STEPPER_STEP_BOTH_EDGE | Result |
|---|---|---|---|
| 8 KiB | off | on | MCU stays on old firmware |
| 28 KiB | on | on | MCU stays on old firmware |
| 28 KiB | off | on | MCU stays on old firmware |
| 28 KiB | off | off | MCU stays on old firmware |

The last row matches the exact settings shown in the screenshot on
issue #105, which is why I'm confused it isn't working for me.

**Vector-table reset-handler comparison** (checked out commit
`7a723bdc` locally to build with byte-identical source to the known-good
`.bin`, to isolate config-only differences):

```
known-good (misc/klipper-v0.13.0-371-g7a723bdc.bin): dd d5 00 08  -> 0x0800d5dd
my build, same commit, 8 KiB config:                 29 85 00 08  -> 0x08008529
my build, same commit, 28 KiB config:                29 d5 00 08  -> 0x0800d529
```

28 KiB gets much closer to the known-good reset vector than 8 KiB does,
which is why I believe 28 KiB is the correct offset for this board —
but the flash still silently fails to take.

## Diagnostic notes

- No `dfu-util`/USB flashing is possible on this board (`No DFU capable
  USB device available`) — confirmed this is expected, since the CH340
  is a plain UART bridge with no native DFU support. SD card is the only
  path.
- I have not been able to confirm whether the bootloader ever
  registers/renames the file after a flash attempt (some other Sovol
  users report the file gets renamed, e.g. to `.CUR`, on a successful
  flash) — on my board the filename is unchanged after every attempt,
  which suggests the bootloader isn't picking up the file at all, not
  that it's writing it incorrectly.

## Questions

1. Does anyone have a **confirmed-working `.config`** (or the actual
   `menuconfig` diff) that was used to build the `v0.13.0-371-g7a723bdc`
   binary currently shipped in `misc/`? I'd like to compile against
   that exact config as a sanity check before introducing any other
   variable.
2. Has anyone successfully self-compiled and flashed a *newer* firmware
   on this exact board revision (`SV_NL V1.3.1`)? If so, what toolchain
   (`gcc-arm-none-eabi` version) and full `.config` did you use?
3. Is there a way to verify from the host side whether the SD-card
   bootloader on this board actually attempted to read/write the file
   (e.g. a status LED pattern, a log left on the card, anything)? Right
   now I have no visibility into whether the bootloader even runs.

Happy to attach my full `.config` or `klippy.log` if useful.