bassamanator/Sovol-SV06-firmware #108: Update guide & beep macro to work with sv06 screen

frostius, 2023-11-20:
I flashed the screen to work with klipper and discovered that the beep macro does not work.  The M300 gcode is supported by the firmware, we just need to update the beep macro to match.

It would be cool if you could update the instructions to reflect that the beep does in fact work, and also update the `cfgs/misc-macros.cfg` beep macro to use the M300 gcode.

A couple of things to note:
- the firmware throws an error if duration > 255 and aborts the print.  Not cool.
- setting the frequency doesn't appear to do anything.  I haven't verified, but I'm assuming the screen only has a buzzer and not a piezo speaker.

This is a beep macro that includes error handling for duration values > 255 to avoid aborted prints

```shell
[gcode_macro BEEP]
description: BEEP I=3 DUR=200 FREQ=2000: Beep 3 times, for 200ms each, at 2kHz frequency.
gcode:
    # Parameters
    {% set i = params.I|default(1)|int %}           ; Iterations (number of times to beep).
    {% set dur = [params.DUR|default(100)|int,255] | min %}     ; Duration/wait of each beep in ms. Default 100ms.  Cap at 255 to avoid error
    {% set freq = params.FREQ|default(2000)|int %}  ; Frequency in Hz. Default 2kHz.

    {% for iteration in range(i|int) %}
        M300 P{dur} ;send to screen
        G4 P{dur}
    {% endfor %}
```

frostius, 2023-11-21:
> You're using the new Sovol klipper pad, or are you using the original
> screen of the plus?
>
> I'm curious about your version of Klipper, because M300 isn't a command
> in klipper.

Original non-klipper screen.

M300 is supported by the screen firmware per the firmware GitHub site

frostius, 2023-11-21:
Flashing the screen requires flashing an updated version of clipper to the
printer. I'm guessing M300 support is included in that branch?

bassamanator, 2023-11-21:
You're talking about the firmware from https://github.com/fryc88/klipper-sv06plus-screen?

If that is the case, then no, `M300` is not a built-in Klipper command, source: https://www.klipper3d.org/G-Codes.html?h=m11#g-code-commands. Must be something related to fryc88's fork of klipper, which is quite outdated at this point I notice now, unfortunately for plus owners.

frostius, 2023-11-21:
> From "Bassam" ***@***.***>
> To "bassamanator/Sovol-SV06-firmware"
> Cc "Matt Frost" ***@***.***>; "Author" ***@***.***>
> Date 11/20/2023 7:31:25 PM
> Subject Re: [bassamanator/Sovol-SV06-firmware] Update guide & beep macro
> to work with sv06+ screen (Issue #108)

One more link deeper to the source: 
https://github.com/Desuuuu/DGUS-reloaded-Klipper

frostius, 2023-11-21:
I pulled from fryc88's last night which is currently on v0.11.  Version 
0.12 was just released 2 weeks ago per 
https://www.klipper3d.org/Releases.html#klipper-0110.  Per fryc88's 
notes, the github is supposedly setup to auto-merge with the master 
klipper branch - maybe gbkwiatt does testing before publishing.

What am I missing that makes you say this is "quite outdated"?  Is there 
a version between 0.11 and 0.12?

bassamanator, 2023-11-21:
Check the commit history, no recent commits:
- https://github.com/fryc88/klipper-sv06plus-screen/commits/main
- https://github.com/Desuuuu/DGUS-reloaded-Klipper/commits/master

bassamanator, 2023-11-21:
Actually, this is the 'fry' version of Klipper, https://github.com/gbkwiatt/klipper/commits/master, was updated recently.

In any case, not quite sure what to do since Klipper official doesn't support this command, and I have already created and maintain 4 branches on this repo alone. I think this kind of `special` info belongs in a discussion, will think about it later.

frostius, 2023-11-21:
Klipper firmware is pulled directly from 
https://github.com/gbkwiatt/klipper

Date 11/20/2023 8:02:10 PM

frostius, 2023-11-21:
> In any case, not quite sure what to do since Klipper official doesn't support this command,

The `DGUS_PRINT_START` and `DGUS_PRINT_END` commands aren't klipper supported either.  If we want a working screen flashed with gbkwiatt's firmware, we're going to have to use instructions supported by that firmware.  In particular, M300 for the buzzer, and DGUS_PRINT_START/END to trigger screen updates.   

If you don't want to support the screen integration, I'd be happy to create a github page that contains an updated version of `misc-macros.cfg` that work with the gbkwiatt screen firmware

bassamanator, 2023-11-22:
Let's think this through.

Where are `DGUS_PRINT_START` and `DGUS_PRINT_END` called `from`?

frostius, 2023-11-23:
> Where are DGUS_PRINT_START and DGUS_PRINT_END called from?
PRINT_START/END & CANCEL_PRINT, as per the instructions at https://github.com/fryc88/klipper-sv06plus-screen

bassamanator, 2023-11-23:
My friend, best is to actually fork this repo (tell it to copy all branches, then delete all branches except for `sv06-plus`).  Create a new branch, checked out from `sv06-plus`, along the lines of `sv06-plus-stock-screen`.

A fork will make it easier to stay up to date with this repo. I'm not sure how proficient you are with `git` so let me know if you need help with `git` commands.

bassamanator, 2023-11-25:
Closing this issue. Feel free to ping me.

closed 2023-11-25 by bassamanator