bassamanator/Sovol-SV06-firmware #113: [BUG] KAMP doesnt seem to be purging.

iliyan61, 2023-11-23:
**Describe the bug**
There's no purge line when KAMP is enabled even with all the settings I can think of enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Send a print
2. wait for the printer to heat up
3. it starts the bed mesh
4. it finishes the bed mesh and starts printing with no purge

**Expected behavior**
I expect it to lay down a purge line.

**What printer are you using?**
- SV06,

**What kind of device are you using to host Klipper?**
 - OS: [Raspberry Pi]
 - Version: [4b+ on the latest debian]

**How did you install Klipper?**
 - KIAUH, 

**Is your local copy of this configuration up to date?**
- Yes I just installed and set it up today.

bassamanator, 2023-11-23:
What slicer?

iliyan61, 2023-11-23:
Prusa Slicer. I had KAMP working on a previous install on klipper but i did a reinstall earlier today so I haven't touched pursaslicers code. The start G-Code is below I'm sure I've missed something but I'm not sure what it might be. Could you perhaps provide the steps to go through to re-enable it as it's not working across 2 printers and 2 separate machines.

'M104 S0 ; Stops PS/SS from sending temp waits separately
M140 S0
PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_temperature[initial_extruder]]'

bassamanator, 2023-11-23:
You're missing `purge_line`, please see https://github.com/bassamanator/Sovol-SV06-firmware#adjust-your-slicer.

To use KAMP's line purge, please the KAMP faq section.

closed 2023-11-23 by bassamanator

iliyan61, 2023-11-23:
I am fairly sure I have setup KAMP and it's purge line correctly going off what is written in your readme and I have gone through most of the configs related to it and enabled what seems necessary however it still doesn't work. I don't think I need to update my slicer G-Code as i already had it working with KAMP.

I followed your instructions anyway and it just lays a generic purge line. Could you please share the instructions to get KAMP purge line working with your config?

bassamanator, 2023-11-23:
Docs adjusted to add clarity: read kamp faq and adjust slicer sections again. Also pull repo.