bassamanator/Sovol-SV06-firmware #132: Prusa print profiles - name explanation

blackICE19240, 2024-01-04:
I'm pretty new to 3D printing so sorry if I'm expected to know this but could you explain the names of the Ellis print profiles ..

- What does "PIF" (Vs decorative) mean?
- What do the different "_" and "__" prefixes mean?
- What do the 24 and 15mm3 mean? Is it a flow rate like 24mm³/s ?
- Is the "0.20mm QUALITY SV06 Ellis" only for the SV06 and why is it named differently?

bassamanator, 2024-01-04:
> What does "PIF" (Vs decorative) mean?

Think structural parts vs decorative.

> What do the different "_" and "__" prefixes mean?

Allows profiles to be grouped together.

> What do the 24 and 15mm3 mean? Is it a flow rate like 24mm³/s ?

Yes.

> Is the "0.20mm QUALITY SV06 Ellis" only for the SV06 and why is it named differently?

I don't see that, you might not have imported correctly. See the FAQs and Discussions > Q&A for related discussions; open discussions, me as author. https://github.com/bassamanator/Sovol-SV06-firmware/discussions/categories/q-a

blackICE19240, 2024-01-04:
Sorry, I forget discussions is a thing sometimes. Thanks for replying so fast too.

I think I must've picked up that other one somewhere else and sorry again for not noticing the grouping, it seems obvious now.

closed 2024-01-04 by blackICE19240