bassamanator/Sovol-SV06-firmware #133: Help with BTT PAD 7 Adxl345

Degenfoster, 2024-01-05:
Hey sorry i'm quite new to klipper and i just got it to work with my sv06+ with your guide, but i can't find how to use the ADXL 345 for input shaping with the BTT pad7, enabling adxl direct caused an error and klipper doesn't launch, please forgive me I'm really beginner at coding and stuff so i don't really understand it

MacBandito, 2024-01-05:
Did you follow the instructions in the Pad 7 manual?  You should have the BoardEnv.txt file thats installed when you follow the BTT instructions. You may have deleted it if you followed the Bassamanator instructions.  You need that file though to activate the correct setting.   I’ve pasted the instructions from the manual below.  

Pad 7 Settings
Open the "BoardEnv.txt" configuration file, and set the following parameters:

spidev1_1: Enables the spidev1_1 to the system user space, allowing the Pad7's SPI port to connect to an ADXL345 accelerometer module.

Setting up ADXL345
As explained in "Pad 7 Settings" section, set the overlays to include spidev1_1.
After booting, the system user space should load spidev1.1. Add the following
configuration to the printer.cfg file to use the ADXL345:
[mcu CB1]
serial: /tmp/klipper_host_mcu
[adxl345]
cs_pin: CB1:None
spi_bus: spidev1.1
axes_map: z,y,-x # Modify according to the actual orientation of the ADXL345
installed on the printer.

Degenfoster, 2024-01-05:
I think you are right about  the config file being deleted, Is there a way to recover the file?

Degenfoster, 2024-01-05:
You are amazing! The config files of the pad 7 was already good, but the trick was the CB1 I kept following the wrong set of instructions for the CM4 board instead! So sorry rookie mistake but glad I finally could fix it!

MacBandito, 2024-01-05:
You’re welcome.  I’m very new to this too.

bassamanator, 2024-01-05:
@MacBandito Thanks for helping out the community! I've edited your original post for styling and to add sources. Please see if it still all makes sense or if I've blundered something.