bassamanator/Sovol-SV06-firmware #141: Add a mcu temperature sensor section

graccoon, 2024-01-22:
Fortunately, stm32 has an internal temperature sensor, which allows you to track frequent errors with its overheating (don't forget that in the standard configuration the MCU cooling fan is connected to the model cooling fan). In particular, the famous "timer too close" error has been haunting me for the last couple of weeks, and thanks to working with cooling and temperature tracking it was possible to eliminate MCU overheating.
And the proposed section is very simple:
```
[temperature_sensor mcu_temp].
sensor_type: temperature_mcu
min_temp: 0
max_temp: 100
```

graccoon, 2024-01-22:
I've found source of my code block:
https://klipper.discourse.group/t/stm32h743xx-mcu-temperature/5867/2

bassamanator, 2024-01-23:
This is nothing new; feel free to look through closed issues on this. The reason this section was *removed* from this config was because it doesn't play nicely with this motherboard. 

It's very likely your issues are related to having added this section.

closed 2024-01-23 by bassamanator