bassamanator/Sovol-SV06-firmware #155: Mention Blue Screen after flashing Klipper

TLDlegal, 2024-04-07:
Could you possibly mention in the instructions that it is completely normal after flashing Klipper for the first time the screen will stay blank till you connect to it with the raspberry pi?
I flashed it a hundred times and thought it failed but i read days later on a random reddit comment that you have to set the correct port in the pi first so it can connect and finally show up some text.
I was waiting for some kind of sovol logo or Klipper.

bassamanator, 2024-04-07:
In the Flash Firmware section, it says:

> ⏲️ At this point, it's not possible to tell with certainty whether your flash was successful, continue on with the guide.

This is a **fact**. Why did you not follow these very clear instructions?

If you can convince me that you would follow the *proposed* instructions, when you did not follow the *existing* instructions, I will make the change you requested, because at this point, and I'm not trying to be harsh here, you seem like an instruction-ignorer.

Please think on the matter before you respond.

TLDlegal, 2024-04-07:
You misunderstand me. I followed the instruction word for word, but as someone that has not seen a Klipper UI before i was just waiting for the old. Interface. I thought it. will look the same and changes are only made in the background. If i only see a blank screen i automatically think it failed.

bassamanator, 2024-04-07:
> You misunderstand me. I followed the instruction word for word

My friend, how can you make that claim, when the instructions state:

> ⏲️ At this point, it's not possible to tell with certainty whether your flash was successful, **continue on with the guide**. [emphasis added]

Please think on the matter before you respond.

\* *Ref A*

TLDlegal, 2024-04-07:
You are right, i could have followed the instructions blind. But my suggestion was to ad a hand full of words. Its your choice and the guide is wonderful, even i as a beginner managed to do it. But i flashed a few smartphones and consoles and blank screens automatically make me wonder what i did wrong.

For example like this:

_⏲️ At this point, it's not possible to tell with certainty whether your flash was successful; you'll see a blank screen, and it is normal. Continue on with the guide._

Jonybat, 2024-04-07:
Having gone through this process recently, I must say that I understand where the op is coming from. 

Even though the instructions say its not possible to know for sure if it was successful or not, and to continue as normal, I do agree that adding a simple mention that the screen will stay blue until later on, will help give more confidence that we are on the right track.

bassamanator, 2024-04-07:
I have worked very hard, completely for free, to ensure that the instructions are 100% accurate, and as concise as they can be, because as demonstrated by this issue, people don't follow instructions (again, not trying to be harsh, just stating facts). So I'm back to square one in my head: 1)`How can more instructions be helpful to people who don't follow instructions?` and 2)`At how many more places must I add more words and sentences, to hand hold users who don't follow instructions?`

@TLDlegal @Jonybat you guys might not be aware, but this guide, instead of being a 9 minute read, was twice as long, if not longer. I decided many months ago to `reduce the size of this README` while still remaining `completely accurate`. And guess what? The number of frivolous issues and discussions has gone almost down to zero (serious). This is clear indication to me that I'm on the right track. Do I really want to go back to my old way of thinking, to hand-hold users (which is my natural state), as they progress through the `README`, if it leads to poorer outcomes? Absolutely not. BTW, I've hand-held users with this `README` on live audio/video chat completely for free, all around the world (one user that I helped was in Japan); just search for `talky.io` in the discussions for some evidence of this.

If I add this change
> ⏲️ At this point, it's not possible to tell with certainty whether your flash was successful; you'll see a blank screen, and it is normal. Continue on with the guide.

then a documentation writer such as myself will have to make many other changes, leading to a larger `README`. Perhaps this is a failing of mine?

The only way I see to deal with point 2 above is to change:
> Follow the steps in order. If an error was reported at a step, do no proceed to the next step.

to
> Follow the steps in order. If an error was reported at a step, do no proceed to the next step. ⚠️ Do not try to out-think the instructions.

Something along those lines.

Having said all that, **you guys do raise a valid point**, because I myself considered adding the note about the `blank screen` long before it was raised by anyone else. But again, I cannot be `more` clear, `more` concise, and `more` accurate then `⏲️ At this point, it's not possible to tell with certainty whether your flash was successful, continue on with the guide.` I may certainly lose the battle with a couple of users here or there, but I am winning the war, I assure you.

Does this help you guys see things from my perspective?

Jonybat, 2024-04-07:
Yes, I understand where you are coming from. It's all about the target audience, and it's up to you to decide that.

My MO (and I bet I am not alone) is to usually skim through the docs and jump to the part that I am looking for, and end up missing some important detail in one paragraph, that the author did not think of highlighting.

However, seeing that I did not want to brick my new printer, and that you seemed to have a really nice and concise writeup, I read it all (before I even decided to go through with it) and that part was the only one that left me with a bit of uncertainty. Judging by the wording, I assumed that there would be no indication of the flashing process, and that the printer would not work until some later step, but that was my interpretation. 

If you mention to stop at errors, you will get someone saying they stopped because the screen was blank. The sovol flashing video you linked is about the stock firmware flashing, so the visual output will be different. 

IMO, if you want to be clear, concise and accurate, the best way is to describe the expected state with the least amount of words. Example:
```
... 
5. Turn on the printer and wait a minute (usually takes 10 seconds). The printer screen should stay blank indefinitely 
6. Turn off the printer and remove the microSD.

At this point, it's not possible to tell with certainty whether your flash was successful, continue on with the guide.
... 
```
If I remember correctly from my attempt, it was so that the Sovol logo kept coming up right away after step 5. I found it suspicious and searched the discussions and maybe Google, and found someone mentioning that the logo is not supposed to come at all (which then shifted my attention to the sd card). So I think that adding that mention not only adds reassurance that the reader is following the steps correctly, but it also tells when something is wrong. 

Does this make sense?

bassamanator, 2024-04-08:
> ... The printer screen should stay blank indefinitely

Unfortunately, this might not be the case. The instructions are written in a way that you can stop (fail) at any point, and start with the same exact instructions. It is possible for the user to mistakenly come to the conclusion that they have erred after the MCU path adjustment through sheer happenstance. When they resume at the flashing process, depending on the linux distribution in use, the screen actually might not remain blank.

> I read it all (before I even decided to go through with it) and that part was the only one that left me with a bit of uncertainty.

`uncertainty`, yep, I get it.

You've convinced me that I need to give this issue more thought. I will keep this issue open, and tackle it with a fresh mind in a couple of days/weeks. I might not come to a conclusion that appeases you, but I will certainly give it serious thought.

Thanks to both of you for your input. Feel free to add to the conversation if it pleases you.

JosephBrooksbank, 2024-05-07:
For what its worth, the first microSD card I used **failed** to flash entirely, and I was greeted with the marlin boot screen. I'm familiar with the flashing process so I knew to grab another SD card until I got a blank screen, but for someone who is less familiar they might see the original marlin screen and think that 

> At this point, it's not possible to tell with certainty whether your flash was successful, continue on with the guide.

means "the marlin screen is intended, continue"

Thank you for your work on this repo, btw! Extremely helpful.

bassamanator, 2024-08-29:
Added `important` markdown notice to that section. Thanks all. 🙏🏻

closed 2024-08-29 by bassamanator