bassamanator/Sovol-SV06-firmware #167: [BUG]  CANCEL_PRINT can collide with parts on the bed

Cornersruns, 2024-05-05:
**Describe the bug**
CACEL_PRINT raises the gantry slightly then moves the bed all the way back.  If there is a part at the front of the bed that is higher than the Z Height of the print the head collides with the part.  This is especially true when Print Sequence is by object.

**To Reproduce**
Steps to reproduce the behavior:

1. Start a print with Print Sequence by object, place a tall part at the front of the bed.
2. Once the tall part is printed and the printer has started on another item cancel Print
3. The print bed moves backward without the head being raised high enough to clear it.
4. The print head collides with the first printed part

**Expected behavior**
On cancel print, raise the print head but do not move the bed.

**What printer are you using?**

SV06 Plus,

**What kind of device are you using to host Klipper?**

Raspberry Pi

**How did you install Klipper?**

KIAUH

**Is your local copy of this configuration up to date?**
Yes

- This configuration changes all the time, when did you install it? Are you sure you're running the latest version?

**What slicer are you using?**

- [ ] PrusaSlicer
- [ ] SuperSlicer
- [ ] Cura
- [X] OrcaSlicer

**Are you using the config bundles from this repo?**

- [X ] Yes
- [ ] No

**Did you [Adjust your slicer](https://github.com/bassamanator/Sovol-SV06-firmware#adjust-your-slicer)**?

- [ X] Yes
- [ ] No

**Additional context**
Add any other context about the problem here.

bassamanator, 2024-05-06:
Before i think about this, how did you import the ps/ss config bundle into OS? Does OS have this functionality now?

Cornersruns, 2024-05-06:
I'm using Christian Vicks Orca Slicer Presets V1.5.5. With this start g-code:

`M104` S0 ; Stops PS/SS from sending temp waits separately
M140 S0
PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_temperature[initial_extruder]]

`LINE_PURGE`

Klipper configs are purely yours with no other additions.

The cancel print is sent from klipper, Orca Slicer has a slicing setting of by Object so that each object, if multiple objects on a plate, is fully printed before the next object is started (instead of printing the whole plate layer by layey)

bassamanator, 2024-05-15:
Thanks for bringing this up. Unfortunately, I don't see how I can pull a previously printed parts height to include in the `print_end` macro, which is invoked during the cancellation process.

A `Known Issues` section has been added to the `README`; f27b569d0c16be6226833a225ea0535b65409d7c.

closed 2024-05-15 by bassamanator