bassamanator/Sovol-SV06-firmware #169: SV06 SKR mini E3 V3 runout sensor

Astrolava, 2024-05-11:
I hooked up a filament runout sensor to the toolhead of my SV06, which uses an SKR Mini E3 V3, I'm very sure it's wired correctly.
My printer.cfg has the following: 

[filament_switch_sensor filament_sensor]
switch_pin: !PA4 # "Pulled-high"
pause_on_runout: True
insert_gcode:
    M117 Insert Detected
    { action_respond_info("Insert Detected") }
runout_gcode:
    M117 Runout Detected
    { action_respond_info("Runout Detected") }
    CONDITIONAL_BEEP i=3 dur=300 freq=400
    
    I can see there's a "filament sensor" part on Klipper's dashboard now, but it says "empty" no matter if there's filament loaded or not.
    I don't seem to have any indication of a state change at all, and I'm wondering if I missed a step? 
    It's a brand new filament runout sensor, but I don't even know how to test it to make sure the actual piece of hardware works.
    
The filament runout sensor is a cheap one off alieepress, so it's not at all impossible I got a faulty one.
I've tested the wire to confirm all 3 connections work.  

Just to clarify the wiring.  On the toolhead, V is top, G is middle, S is bottom.
On the runout sensor, input side is V, middle is G, output side is S.

Any advice on how I might get this working?

jakebullet70, 2024-05-12:
I just setup one on an SKR 1.4 board on another printer. I used these docs.

https://www.klipper3d.org/G-Codes.html#filament_switch_sensor

I used 'QUERY_FILAMENT_SENSOR' command from the terminal until I had it working. For me it was the wrong pin.

Astrolava, 2024-05-12:
> I just setup one on an SKR 1.4 board on another printer. I used these docs.
> 
> https://www.klipper3d.org/G-Codes.html#filament_switch_sensor
> 
> I used 'QUERY_FILAMENT_SENSOR' command from the terminal until I had it working. For me it was the wrong pin.

What pin did it end up being for you?
I have just finished testing pins PA-PF (0-15) - aside from the reserved and already used pins.
None of the pins detected the filament runout button being toggled.

I wonder if maybe there's an incorrect pin set somewhere which is occupying the pin for the sensor.  
I noticed my screen's button doesn't work on PA15, but, it also didn't detect the filament runout.
I'm not too sure what to do now.

Astrolava, 2024-05-12:
These are the already occupied pins which I can't test for obvious reasons.  I wonder if one of these might be the actual probe pin, and if so, what that pin might otherwise be.  

PA0 - [extruder] - sensor_pin
PA9 - [display] - encoder_pins
PA10 - [display] - encoder_pins
PA11 - reserved for USB
PA12 - reserved for USB
PA15 - [display] - click_pin (tested)

PB0 - [stepper_z] - step_pin
PB1 - [stepper_z] - enable_pin
PB2 - [stepper_y] - dir_pin
PB3 - [extruder] - step_pin
PB4 - [extruder] - dir_pin
PB5 - [output_pin beeper] pin
PB8 - [display] - cs_pin
PB9 - [display] - sclk_pin
PB10 - [stepper_y] - step_pin
PB11 - [stepper_y] - enable_pin
PB12 - [stepper_x] - dir_pin
PB13 - [stepper_x] - step_pin
PB14 - [stepper_x] - enable_pin
PB15 - [controller_fan electronic_enclosure_fan] - pin

PC0 - [tmc2209 stepper_x] - diag_pin
PC1 - [tmc2209 stepper_y] - diag_pin
PC4 - [heater_bed] - sensor_pin
PC5 - [stepper_z] - dir_pin
PC6 - [fan] pin
PC7 - [heater_fan hotend_fan] - pin
PC8 - [extruder] - heater_pin
PC9 - [heater_bed] - heater_pin
PC10 - [tmc2209 extruder] tx_pin
PC11 - [tmc2209 extruder] uart_pin
PC14 - [probe] - pin

PD1 - [extruder] - enable_pin
PD6 - [display] - sid_pin

PF0 - reserved for crystal?
PF1 - reserved for crystal?

Astrolava, 2024-05-12:
I guess another option here might be that the place where the P7 is plugged in requires more configuration.
If the port on the board, which I believe is labeled "DET (power detection?)" detects power being input, then it might require a source of power (3v or so?)  - but, if the DET doesn't also supply that power, then the filament sensor - which is just a button that joins two wires essentially - is just a dead wire.
Doesn't the output (DET) need to supply 3v to Vcc, then detect 3v in ground?  Is that how it works?

Right now, it seems that it's just a dead wire.  I couldn't get any reading on any of the pins of P7.

bassamanator, 2024-05-12:
You need to uncomment the filament sensor section in `printer.cfg`, as well as adjusting `misc-macros.cfg`. Please see [What are some settings that I can change?](https://github.com/bassamanator/Sovol-SV06-firmware/blob/skr-mini-e3-v3/README.md#what-are-some-settings-that-i-can-change). Read full docs before proceeding.

Ignore what it says on the dashboard. Start a print with no filament in the nozzle, but a piece of filament in the sensor. Remove the filament from the sensor to see if things are working as expected.

Astrolava, 2024-05-13:
> You need to uncomment the filament sensor section in `printer.cfg`, as well as adjusting `misc-macros.cfg`. Please see [What are some settings that I can [...]

Am I still doing the pin PA4 ?

Astrolava, 2024-05-13:
So yeah, after testing and testing, it doesn't stop the print.  Doesn't pause it, doesn't beep, no indication at all that anything has changed.
It'll just keep printing without filament.

jakebullet70, 2024-05-13:
Had the same issue. You need to turn it on in the START gcode.

Astrolava, 2024-05-13:
> Had the same issue. You need to turn it on in the START gcode.

Thanks for the reply.  Umm, sorry I actually don't know how to turn it on.  Mind elaborating?

jakebullet70, 2024-05-13:
Yes that is right: SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=1

When setting up a filament sensor you do not need to print, you can use 'QUERY_FILAMENT_SENSOR' command and put filament in the sensor by hand, that tells you if you have the correct pin setup, once that is done and QUERY_FILAMENT_SENSOR works by telling when it has filament or not then you turn it on in your START gcode.

bassamanator, 2024-05-13:
> Am I still doing the pin PA4 ?

@Astrolava revert to whatever was already in `printer.cfg`.

Do not add `SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=1` but if you think you know better than the author of this config, go for it.

jakebullet70, 2024-05-14:
I sad that is what  had to do on an SKR 1.4 board. Did not work with out it.  Sorry,  I wll back out.

bassamanator, 2024-05-14:
@jakebullet70 No need to back out. The issue is that you'll have a hard time keeping up with any changes in this config. The only thing my users are required to do are to either uncomment/comment out a section, or change a 1 to a 0 or vice versa. No programming knowledge is *required*.

bassamanator, 2024-05-15:
Updated wiring diagram, printer.cfg and misc-macros.cfg. Pull new changes.

1. Take note of the wiring connecting to the SKR's `e0-stop`
2. Take note of how the filament sensor is connected to the hotend pcb.
3. Take note here: https://github.com/bassamanator/everything-sovol-sv06?tab=readme-ov-file#filament-sensor
4. Enable filament sensor via: https://github.com/bassamanator/Sovol-SV06-firmware/blob/skr-mini-e3-v3/README.md#what-are-some-settings-that-i-can-change

Can't simplify the process any further.