bassamanator/Sovol-SV06-firmware #176: KAMP has stopped  generating a mesh only in the area of printing

bladyle, 2024-08-12:
Hi, I'm using your klipper configuration since beginning and I'm very happy with it but since yesterday I noticed that when I send file to print and with activated KAMP, bed mesh is generated for the whole bed - not only for the object. 
The only thing that I have modified to the system is that I casually run KIAUH  scripts for updates. 

**What printer are you using?**

- SV06

**What kind of device are you using to host Klipper?**

- OS: old laptop
- Version: []

**How did you install Klipper?**

- Manual install, KIAUH

**Is your local copy of this configuration up to date?**

- This configuration changes all the time, when did you install it? Are you sure you're running the latest version?

**What slicer are you using?**

- [ ] OrcaSlicer

**Are you using the config bundles from this repo?**

- [ ] Yes
- [ ] No

**Did you [Adjust your slicer](https://github.com/bassamanator/Sovol-SV06-firmware#adjust-your-slicer)**?

- [ ] Yes
- [ ] No

bassamanator, 2024-08-12:
How did you enable KAMP? Is the configuration up to date with the repo?

bladyle, 2024-08-14:
I enable it from cfgs/misc-macros.cfg but I have seen now that are some new changes to repo. 
I will update it and test again. Will report back when I test it.

bassamanator, 2024-08-14:
~Well, if you truly are up to date with the repo, you could try flashing a newer `klipper.bin`. This really shouldn't be necessary at all, but given the fact that~

Now that I think about it, it might be possible to determine if you're up to date with the repo. Run `_BED_MESH_CALIBRATE` in your klipper console (make sure your build plate is clear). What happens?

bladyle, 2024-08-15:
`_BED_MESH_CALIBRATE`   created new default mesh, although I noticed couple of `Probe samples exceed tolerance. Retrying...` when doing it

bassamanator, 2024-08-15:
The logic for the kamp routine is quite simple, even a non-coder can likely make sense of it,
```
    {% if kampEnabled == 0 %}
        BED_MESH_PROFILE LOAD=default                        ; NOTE if not using a mesh, comment out this line
        M104 S{hotendtemp}                                   ; set & don't wait for hotend temp
        G28 Z                                                ; final z homing
    {% else %}
        G28 Z                                                ; final z homing
        BED_MESH_CALIBRATE                                   ; KAMP mesh
        M104 S{hotendtemp}                                   ; set & don't wait for hotend temp
    {% endif %}
```
Given that Kamp works for me, I suspect that you either have not enable it, or your configuration isn't fully up to date.

I recommend you re-download the configuration (do the git method if you want to make your life easier with managing your config).

**Also, to clarify, KAMP was previously working for you?**

bladyle, 2024-08-15:
I have edited my last comment - it was my mistake...
I will use the post to ask you for another notification I receive now in klipper but didn't receive it before:

> Stepper driver warning: OTPW flag set
> 
> The stepper driver 'stepper_z' has triggered the OTPW flag and may stop working if it gets any hotter. This is an indication of an over temperature condition. This can be caused by a too high current. Please check the stepper driver settings and cooling.