bassamanator/Sovol-SV06-firmware #191: fichier moonraker.

yan3700, 2024-11-15:
bonjour
j'ai pris un des dossier moonraker.conf je les mis dans mon mainsail apres j'ai recuperer sa 
[update_manager printer_config]
type: git_repo
primary_branch: master
path: ~/printer_data/config
origin: https://github.com/bassamanator/Sovol-SV06-firmware.git
managed_services: klipper

est depuis je peux plus l'enlever est je peux plus modifier mais fichier est plus sauvegarder je vous fais une capture d'ecran 
[image]

bassamanator, 2024-11-16:
Regardless, from what i see in the picture, this is a bad idea.

yan3700, 2024-11-16:
depuis que j'ai rajouter c'est lignes de la lignes44 a la ligne 49 j'arrive pas a les supprimer c'est partie la je les prit dans un commentaire ou il parler des mis a jour est depuis je suis en lecture seuil comme indiquer au dessus a gauche j'arrive plus a modifier tous les fichier depuis

bassamanator, 2024-11-16:
So you're unabled to edit `moonraker.conf` anymore?

What I would do is to edit `moonraker.conf` in the terminal.
1. SSH into the device.
2. `cd ~/printer_data/config`
3. `nano moonraker.conf` (or use `vi` to edit)

Furthermore, unless you have made very intricate changes to the configuration, why not just start from scratch, meaning, delete the contents of `printer_data/confg` and begin again.

yan3700, 2024-11-16:
justement je voulais repartir de zéro avec une Raspberry Pi 4 que j'ai commander mais comment fais vue qui y'a dejas cette installation sur le pc est sur ubuntu y'a aussi les fichier je c'est pas comment faire

bassamanator, 2024-11-16:
1. Flash raspberry pi os onto a microSD card
2. Install KIAUH
3. ...

https://www.youtube.com/watch?v=jGplZnNoT4I

yan3700, 2024-11-16:
Ok merciest sa va remettre après le boîtier klipper en état c'est à dire tous les fonctionnements que j'avais aux débuts [image]

J’ai celui  là c’est klipper ou klipperscean que je doit installer 
[image]

yan3700, 2024-11-16:
Après j’ai tous mais fichier d’origine que j’ai garder je fais comment pour les mettre à la place se que sont sur mainsail quand je l’aurais installer avec le pi

bassamanator, 2024-11-16:
To reset/update Sovol klipper screen, find instructions here: https://github.com/3DPrintDemon/How-to-Update-Sovol-Klipper-Screen-To-Latest-Klipper-SV06-and-SV07

To install my configuration to the Sovol screen is more complicated. It would take me about 2 minutes, but if you don't know linux, then you'll have a hard time.

Don't forget to star my repo. Thanks.

yan3700, 2024-11-16:
C’est quoi le lien au dessus fait installer et ou ?

yan3700, 2024-11-16:
Après je suis peux être pas obligé de reset le boîtier klipper il peux se mettre quand j’aurai installé tous est ajouter l’imprimante dans mainsail

yan3700, 2024-11-17:
j'ai fais une installation propre avec raspberry j'ai une erreur que j'arrive pas a enlever 
[image]
est pour l'écran il les toujours comme sa depuis la nouvelle installation je suis perdu la  

https://private-user-images.githubusercontent.com/187410033/386874520-726af977-9680-422d-9e8a-b4d8f1cf1295.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE4ODY4NTEsIm5iZiI6MTczMTg4NjU1MSwicGF0aCI6Ii8xODc0MTAwMzMvMzg2ODc0NTIwLTcyNmFmOTc3LTk2ODAtNDIyZC05ZThhLWI0ZDhmMWNmMTI5NS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExN1QyMzM1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MTVmMDY1YmQ4MmJiNDllY2VkMzNjMDBhZGQ2MTRlMDRmMjZlYTVkYmYxODhjZDY0YmI5ZmU0YmU4ODc0ZDlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9._n5XP5ztaO-knn4jzU3fgbN4gZUSoGep8LKvINhC_sk