bassamanator/Sovol-SV06-firmware #202: [BUG]Beeper

Bandit-Designs, 2025-01-28:
**Describe the bug**
sv06 beeper is on all the time

**To Reproduce**
turn up beeper you can hear it 

1. Go to 'beeper'
2. Click on 'beeper volum.'
3. Scroll down to 'beeper in desktop'
4. See error

**Expected behavior**
a screaming incredibly annoying beeeeeeeeeeeeeeeeeeeeeeeeeeepeeeeeeepeeeepeeeppee

**Screenshots**
If applicable, add screenshots to help explain your problem.

**What printer are you using?**

- SV06,.

**What kind of device are you using to host Klipper?** Laptop

- OS: [e.g. old laptop, PC, Raspberry Pi]
- Version: []laptop all up to date software

**How did you install Klipper?**with linix

- Manual install, KIAUH, MainsailOS, FluiddPi, etc. Mainsail 

**Is your local copy of this configuration up to date?**as of tonight again

- This configuration changes all the time, when did you install it? Are you sure you're running the latest version?
always keepm,it up to date 
**What slicer are you using?**
Prusa, 
- [ *] PrusaSlicer
- [ ] SuperSlicer
- [ ] Cura
- [ ] OrcaSlicer

**Are you using the config bundles from this repo?**

- [* ] Yes
- [ ] No

**Did you [Adjust your slicer](https://github.com/bassamanator/Sovol-SV06-firmware#adjust-your-slicer)**?

- [** ] Yes
- [ ] No

**Additional context**
Add any other context about the problem here.
I need a beeper that doesn't drive my dogs to jump through windows

bassamanator, 2025-01-28:
Go over the **steps to reproduce** again please.

Bandit-Designs, 2025-01-28:
All I do is turn up the beeper on the klipper screen below the fan slider and the beep starts. I have to leave the beep volume turned off completely or my dogs go insane lol.
It’s been like this for awhile now. I’ve been searching for a fix for a long time. I just uploaded the latest from GitHub and it still does it